iic-jku/analog-circuit-design #8: Info boxes w/ Latex equations

michaelk99, 2024-08-22:
Latex equations inside quarto info boxes seem not to render correctly.

Examples:
Middlebrook's Method
Integral for evaluating total noise of the mos-diode: only equation is visible

hpretl, 2024-08-24:
Middlebrook is fixed, one needs to have an empty line before the closing `:::`. The noise integral is also fixed by having an empty line above and the below the equation. Good find, we need to be careful when using the callout boxes

hpretl, 2024-08-24:
OK, another interesting issue: The callout box (Middlebrook) worked fine when rendering just this one page, but when rendering the complete document it does not look correct. I now moved this into a dedicated appendix, is any way better, I think.

closed 2024-09-14 by hpretl