iic-jku/analog-circuit-design #25: Use Tian's method instead of Middlebrook's?

hpretl, 2024-11-26:
Currently, the course uses Middlebrook's method for loop gain analysis. A slightly better method would be Tian's (M. Tian, V. Visvanathan, J. Hantgan, and K. Kundert, “Striving for small-signal stability,” IEEE Circuits Devices Mag., vol. 17, no. 1, pp. 31–41, 2001, doi: 10.1109/101.900125), so maybe change in a future revision.

simi1505, 2024-11-26:
Recently, I found an interesting paper on ten different simulation-based loop gain analysis methods: https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=6977998

simi1505, 2024-11-30:
I have added Tian's method to mosfet_diode_loop.sch and compared it with Middlebrook's method. Interestingly, both loop gains are equal at certain points as expected. However, the calculated result of ft=23GHz is simulated more precisely with Middlebrook's method instead of Tian's method which is not expected, right? Or maybe the calculated ft is already too high?

Maybe someone could double-check my simulation. As far as I understood the procedure: one draws again two circuits, measures A, B, C, and D and calculates T.
If my procedure is correct, I could also add it to the 5T-OTA and the improved 5T-OTA.

Just in case someone wants to try Middlebrook's and Tian's methods in LTspice, there are example circuits in the examples/Educational folder (LoopGain.asc = Middlebrook's method, LoopGain2.asc =Tian's method).

My literature research resulted in three main advantages over Middlebrook's method:
- accounts for reverse feedback (it doesn't even matter if you reverse the direction of the probe, you still compute the same loop gain)
- the inserted probe elements result in a smaller, sparser circuit matrix
- should result in higher accuracies at the transient frequency of the loop gain

hpretl, 2024-12-03:
Now Tian's method is added and compared to Middlebrook's. Let's close this.

closed 2024-12-03 by hpretl