iic-jku/analog-circuit-design #38: Typo error

ShravanIISC, 2025-02-16:
Dear Sir,

I thank you and your team wholeheartedly for this great tutorial.

I came across this typo error in section  17.1 Open-loop Configuration.

vin,p=vin,p = 0 V.

[image]

simi1505, 2025-02-20:
Hi @ShravanIISC!

Thanks for pointing out this typo. I fixed it and made a pull request.

It would help us even more if you could make a PR yourself the next time.

Thanks again and best regards

Simon

simi1505, 2025-02-20:
@hpretl I think this issue can be closed. I don't have permission to do that.

hpretl, 2025-02-22:
Closed as fixed and merged. @ShravanIISC thanks for the contribution!

closed 2025-02-22 by hpretl