iic-jku/analog-circuit-design #39: MOSFET Testbench

EngGhaith, 2025-02-19:
Hi everybody!
I have modified both NMOS/PMOS schematic to be this way:

[image]

I have only changed the values for L, VGS, VDS.

The simulation is running non-stop for more than 8 hours.
Is it OK?
Or did I make any mistake?

Where should I find the result file then: techsweep_sg13_lv_nmos.txt?

Thank you very much in advance.
Ghaith.

simi1505, 2025-02-20:
Hi @EngGhaith!

This will take longer than the techsweep before, yes, but 8 hours seems too long for me. How long did the default techsweep take?

Maybe it is a problem to increase Vgs / Vds to 2.4V over the max. voltage rating of the transistor?

As a reference, I also added some lengths (15u, 20u and 40u). This increased the run time but was finished within 30mins (max. 1h) on my normal work notebook.

The `techsweep_sg13_lv_nmos.txt` should be saved in the folder, where you started Xschem. If you started Xschem in foss/designs, it will be there. If you started Xschem in foss/designs/YourProject/xschem it should be located in that folder.

EngGhaith, 2025-02-21:
@simi1505 
Thank you for your feedback.
It tooks about 1 day for both NMOS/PMOS in parallel, but then it was finish.
The Transistors are specified in the PDK with about 2V8 maximum VDS (Breakdown voltage).
They should also be able to work with these VDDs.

Thank you very much.
Ghaith.

simi1505, 2025-02-21:
What I also noticed, running these testbenches in parallel on my notebook took in total longer than running them individually.

For example, I have 4 testbenches, LV NMOS / PMOS and HV NMOS / PMOS. If I run them testbench after testbench it is faster compared to starting them simultaneously.

simi1505, 2025-02-21:
What I also noticed, running these two testbenches (NMOS / PMOS) in parallel on my notebook took in total longer (~30min) than running them individually (~ 2 * 11min = 22min) and this gets worse if you run even more testbenches (e.g. LV NMOS / PMOS and HV NMOS / PMOS) in parallel.
However, I think this depends a lot on the used hardware.

hpretl, 2025-02-22:
@EngGhaith @simi1505 The issue with this sweep is that it is only a single MOSFET simulated in quite a few nested loops. So for each data point the netlist needs to be modified (`alter`), then the operating point and ac/noise sweeps are done, and then the next iteration is started. This is quite inefficient, the whole procedure could be run in parallel on all available cores by wrapping this into a script of some sort. However, this characterization run is only needed very few times, essentially only when the transistor models of a PDK change, or when a new PDK is analyzed. Thus, I think the effort to speed up this simulation bench is likely not worth the effort.

Regarding breakdown voltage of the LV MOSFET: $V_\mathrm{GS}$ needs to stay below 1.65V, and $V_\mathrm{DS}$ below 2.0V (NMOS) / 2.2V (PMOS). So yes, theoretically, $V_\mathrm{DS}$ could be run a bit higher than $V_\mathrm{GS}$, but since these are maximum values, these LV MOSFET devices are designed for nominal $V_\mathrm{DD} = 1.5 \text{V} \pm 10 \text{percent}$.

hpretl, 2025-03-10:
@EngGhaith Is there still a question open, or can we close this issue?

EngGhaith, 2025-03-10:
Thank you very much, I think we can close it now.

closed 2025-03-10 by hpretl