iic-jku/analog-circuit-design #41: Clarification on bandwidth calculation in gm/ID sizing

NooriDan, 2025-02-28:
Hi, hope you’re well!

I tried to verify the gm/ID sizing for the basic 5t OTA and i noticed the notebook refers to the bandwidth as the -3dB bandwidth but upon simulations and some inspection I think it is the unity gain bandwidth that is actually calculated. 

I ran the simulation with the sizes in that notebook with xschem + ngspice (using examples in IIC OSIC) and noticed it is the unity gain bandwidth that is equal to 14 MHz. 

I’m not sure if I’m missing something or this was a typo in the notebook. 

Thanks

hpretl, 2025-03-10:
@NooriDan Yes, you are correct. Good observation, I definitely need to fix this.

hpretl, 2025-03-10:
BTW, the confusion here is the unity bandwidth of the OTA is equal to the -3dB bandwidth of the voltage follower/buffer built with it, as it has a gain of $A_v=1$. I will clarify this in the notebook comment.

hpretl, 2025-03-10:
Corrected with 7c6f9dc44906f292d9afafb1e03931487cb28fa2

closed 2025-03-10 by hpretl