iic-jku/analog-circuit-design #45: Docker with Apple Silicon M4

ozgurdenizt, 2025-04-05:
Hi there,

Analog design professional here! Firstly, great work on what you have created.. Here I am using the tools to guide senior EE students at Koc University into analog/mixed-signal design!

While setting up the docker for a student/mentee of mine, who uses Mac M4, although we can get the start_vnc.sh into the docker container, when we click run it immediately terminates. We made sure to follow the steps described in the link "https://kwantaekim.github.io/2024/05/25/OSE-Docker/#vnc" several times, but it seems xquartz and docker cannot get agreed on the display.. Do you have any similar experience before in apple silicon M4? Appreciate any help in advance..

All the best,
Ozgur

hpretl, 2025-04-10:
@ozgurdenizt I also use Macs (M1 and M4) and have no issues. I have installed XQuartz and use the `start_x.sh` script, although VNC mode using `start_vnc.sh` also works. The one important thing is to allow connections from network clients in the XQuartz settings:

[image]

hpretl, 2025-04-16:
Let's close for now.

closed 2025-04-16 by hpretl