iic-jku/analog-circuit-design #47: Small Signal Model and gds

emmersonv, 2025-05-31:
Hello,

When looking at the [MOSFET small signal model](https://iic-jku.github.io/analog-circuit-design/analog_circuit_design.html#sec-mosfet-smallsignal-model), how can gds be modeled as a resistor? Shouldn't it be 1/gds?

If a current source is used to model the dependency of drain current from VDS, then VDS is across the terminals of the current source.
This means the current source can be modeled as a resistor:
ids = vds*gds  ->  vds/ids = 1/gds

If that is the case, would the same apply to the t model? Both gm and gds would be represented by their reciprocals.

Thank you!!!!

simi1505, 2025-06-02:
Yes, $g_\mathrm{ds}$ is the output conductance for both models. You could also write $r_\mathrm{ds} = 1 / g_\mathrm{ds}$ for the output resistance, as you mentioned. The main reason why we use $g_\mathrm{ds}$ is that it is more convenient during small-signal hand calculations.

In fact, this might be confusing in the beginning, if you always learnt Ohm's law with $U = R I $. At least it was for me. :-)
But you will see that using $g_\mathrm{ds}$ is extremely helpful.

emmersonv, 2025-06-02:
That answers my question, thank you so much!

closed 2025-06-02 by emmersonv